UnexpectedMaker/manual_pnp_turntable #3: Thin bottom component tray, maybe thicken it to 0.8mm?

Squantor, 2021-02-08:
I think the component tray has very thin bottom. When I slice the STL in cura it is only two layers. that is a bit on the thin side. Would it be an idea to make it a bit thicker so it is at least three or four layers? I print mostly in PETG and the white filament I have is a more translucent then really white. A few extra layers would help with the opacity, but this is a problem on my side.

Another thing I have noticed but not related to the STL's is that when I slice the component tray with cura, the tool path in cura differs from what is being printed! Really strange, but this is cura's problem. Cura shows the first layer begin printed as one big surface and then makes the component pockets, but in reality the first layer prints quite different with small component pockets and then printing the walls seperate.

Thanks for making this device public, I am printing everything in steps. Black base, white component tray and gray assembly tray. Just waiting for the bearings.

For these big flat prints, I am really glad I installed an ABL sensor on my printer.

rglory, 2021-04-19:
Actually it would be nice to make it 1mm thicker and on top of that it will work with  6805-ZZ bearing which is math cheaper on Amazon but 1mm thicker.

UnexpectedMaker, 2021-04-19:
> Actually it would be nice to make it 1mm thicker and on top of that it will work with 6805-ZZ bearing which is math cheaper on Amazon but 1mm thicker.

The STL files are available for those that want to modify the design to work with different thickness bearings. I chose these because I wanted the unit as low profile as possible.

rglory, 2021-04-19:
I understand. But cheapest bearing for 25x37x6mm I can find on amazon costs $11.49 ($5.75/Item)  while 25x37x7mm costs $14.97 ($1.50 / Item) - quite some difference for 1mm thickness difference.

UnexpectedMaker, 2021-04-19:
Ok, so you can buy the cheaper one and modify the STL file to suit. Or you can buy the bearings from Aliexpress and pay 1/3 of the price. Either way you have options :)