UnexpectedMaker/manual_pnp_turntable #7: Tray scrapes on frame - not enough clearance on bearing

bjonesrock, 2021-03-07:
When the larger bearing is pressed flush into the tray it doesn't have enough clearance on the bottom to spin freely

I think the bottom lip for the large bearing needs to be taller

[image]
[image]
[image]

UnexpectedMaker, 2021-03-07:
Mine is pushed 100% as far as I can push it and I don't see that issue. Maybe just a printer tolerance thing?

bjonesrock, 2021-03-07:
It's possible, I had a lot of issues with the filament I used to print this. I don't know why I decided to use something other than Hatchbox.

I didn't buy the larger bearing from Aliexpress, I found an ebay seller that had the same specs:
https://www.ebay.com/itm/10pcs-6805RD-MR25376-61805-6805N-6805-RS-Ball-Bearing-25x37x6mm-HT2-BB51-GCR15/283611747473?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

It's possible the inner ring on the bearing you have is taller than mine

I'm going to try printing some shims at different heights and see is that fixes it
[image]

If that fixes it I'll attach the STL file

bjonesrock, 2021-03-07:
Using a shim with a height of 0.5mm mostly fixed it for me. I think my tray and/or my frame might be slightly warped which is causing the tray to rub on both the acrylic and the frame.

Attached is the STL file. I tried a bunch of different heights. If someone else has this issue later and 0.5mm isn't the right height for you, just scale the STL file on the Z axis to the size you need.

[0.5mm-shim.stl.zip](https://github.com/UnexpectedMaker/manual_pnp_turntable/files/6096545/0.5mm-shim.stl.zip)

closed 2021-03-07 by bjonesrock